rusefi/hellenNA6_issues #1: AC Compressor Relay Output

BeerMoneyMotorsports, 2021-06-10:
H64 currently has the AC Compressor Relay output(ground) connected to an IN_AUX(GPIOB_0).  We need that changed to an output for ground.

BeerMoneyMotorsports, 2021-06-10:
1j needs to provide a ground the relay when AC os active. So im guessing a low side out will work.

BeerMoneyMotorsports, 2021-06-14:
@Qwerty-OFF did you need anymore info?  i used GPIOH_15 on current board with success.

Qwerty-OFF, 2021-06-14:
[image]

BeerMoneyMotorsports, 2021-06-15:
@Qwerty-OFF please confirm that pin is actually OUT_LOW6_PULLUP and NOT 8.  PIN S12 should be OUT_LOW6_PULLUP which i have confirmed works.

Qwerty-OFF, 2021-06-15:
As a result of the refactoring, the names of the modules pins have changed, but the designation has not changed. We can use any OUT_LOWx to control the AC relay.

mck1117, 2021-06-15:
@Qwerty-OFF did the mapping of MCU pin to function change?

rusefillc, 2021-06-23:
@Qwerty-OFF ^^^ ?

Qwerty-OFF, 2021-06-23:
Done. I will make the changelist after we send the board to production.

closed 2021-07-21 by BeerMoneyMotorsports