rusefi/hellenNA6_issues #5: Options Port Assignments

BeerMoneyMotorsports, 2021-06-15:
@Qwerty-OFF this is to replace https://github.com/rusefi/hellen64_miataNA6_94/issues/50 with concise direction. Please do the following for the H64 options port.  these are in no specific order, please populate whatever pin is best suited for it. Please forgive my lack of knowledge of what the particular I/O should be used on the board.  

1- Delete USB in connector
2- Replace second H- with H+ tied to 1B with fuse
3- Knock Input
4- Boost Control Output(possibly needs a diode like IAC)
5- VVT Output(possibly needs diode like IAC)
6- GND
7- +5v
8- Aux input
9- VR input(vehicle speed?)
10- Flex( digital input should work?)
11 through 16 wbo

stefanst, 2021-06-15:
Remarks:

4. Boost Control should have a clamping diode
5. VVT output should have a clamping diode
8, What is AUX input? Analog? Like oil temp, oil pressure?
9. Vehicle speed on a Miata, to the best of my knowledge, is a square wave signal generated by the speedometer on the instrument cluster pin 2F. Not sure if it's 12V or 5V, but I think it's 5V. So a digital input should work, but this needs verification
10. I think flex needs analog input. The signal gives you a percentage of alcohol in the fuel.

BeerMoneyMotorsports, 2021-06-15:
8- like fuel pressure input or oil pressure
9- unsure on those specifics.  
10- Flex on MS is a digital input but i really do  not know.  @mck1117 had previously noted that it should work on an analog.  maybe use one of the dual inputs that have analog and digital capabilities?

stefanst, 2021-06-15:
10- It looks like the standard Flex sensor is a digital frequency-based sensor, so any digital or analog input should work, as long as it's interrupt-capable (I guess, but I'm no expert). Or maybe the maximum 150Hz is slow enough that we don't even need an interrupt-capable pin and can just check it at least every 3ms or so.

rusefillc, 2021-06-15:
Let's try one more time

closed 2021-06-15 by rusefillc