rusefi/hellenNA6_issues #6: H64 Options port

BeerMoneyMotorsports, 2021-06-15:
@Qwerty-OFF  lets try this again.  Sorry for being confusing.  Please see below for the final info.  Pin position in the connector doesnt matter.  

** Delete USB in connector **  
1-  analog input temperature
2- NC
3- Knock Input  
4- Boost Control Output high current low side (needs a diode like IAC)  
5- VVT Output high current low side (needs diode like IAC)  
6- GND  
7- +5v  
8- analog input  pressure
9- NC
10- digital input(fast)
11- Replace second H- with H+ tied to 1B with fuse   
12 through 16- wbo

Qwerty-OFF, 2021-06-15:
[image]
I think it would be better to place GND instead of NC. What current is required for BOOST and VVT? If more than 1 A, we will need to move at least one of the INJ outputs to OUT_LOWx.
[image]

mck1117, 2021-06-15:
move tach/fan to OUT_LOWx outputs, put VVT/boost on INJ outputs (but keep diodes).

mck1117, 2021-06-15:
> move at least one of the INJ outputs to OUT_LOWx

DO NOT DO THIS

the active clamping on the injector drivers is REQUIRED for injectors, and the OUT_LOWx channels don't have it.  They also won't tolerate the current of an injector.

Qwerty-OFF, 2021-06-15:
Don't panic! I meant OUT_IN5-OUT_INJ8

mck1117, 2021-06-15:
oh, ok.  I think it's probably worth putting the guaranteed low current stuff (fan, tach) on OUT_LOWx, and move vvt/boost to the higher current ones

rusefillc, 2021-06-15:
Regarding fast digital input: PAL stm32 dram plays a role here. we already use GPIOA_6 GPIOB_1 for shat positions. Also GPIOF_11 is already VSS

so whatever pin we use for digital flex cannot be anything 1, 6 or 11

Qwerty-OFF, 2021-06-15:
Please check options port connections, pull-ups and pull-downs
[hellen64_miataNA6_94-b-schematic.pdf](https://github.com/rusefi/hellenNA6_issues/files/6658916/hellen64_miataNA6_94-b-schematic.pdf)
We have an unused Dinput module for digital inputs. Will we keep it for debugging purposes, or will we remove it?

mck1117, 2021-06-15:
@Qwerty-OFF did any mappings from STM32 pin to external pin change? OK if the hardware between changed.

Qwerty-OFF, 2021-06-15:
Yes, there are small changes relative to rev. a like tach, AC relay and maybe something else. If needed i can make changelist after ordering rev. b.

mck1117, 2021-06-15:
@BeerMoneyMotorsports will that be a problem with the few boards out in the wild, now with different mapping? Or are we willing to replace those boards at cost with rev b?

BeerMoneyMotorsports, 2021-06-15:
there are a total of 5 H64 boards.  If firmware update is required on them, which it will be, it is a problem.  I would need to hear from @rusefillc to have his thought on replacing them at cost later on if we kill them.  Is it possible to have different firmware for reva and b using two different pin mapping?

mck1117, 2021-06-15:
> Is it possible to have different firmware for reva and b using two different pin mapping?

this is not worth it for 5 boards

BeerMoneyMotorsports, 2021-06-15:
@Qwerty-OFF any way to get that change list so i can see if we indeed have a problem?

rusefillc, 2021-06-15:
@Qwerty-OFF I believe @BeerMoneyMotorsports is asking for all functions which have changed stm32 pins

@BeerMoneyMotorsports what connection functions do you think could have changed stm32 pins?

BeerMoneyMotorsports, 2021-06-17:
@mck1117 please see the updated schematic.  I think it looks good from a options port pinout but not if those will do exactly what we want.   https://github.com/rusefi/hellenNA6_issues/files/6658916/hellen64_miataNA6_94-b-schematic.pdf

[image]

mck1117, 2021-06-18:
@Qwerty-OFF IN_TEMP, IN_PRESSURE, and IN_DIGITAL need pull up/downs. 

- IN_TEMP should get the same pullup as the clt/iat inputs
- IN_PRESSURE should get a high value pulldown (680k or whatever)
- IN_DIGITAL should get a 2-4k pullup to 5v.

BeerMoneyMotorsports, 2021-06-18:
@mck1117 "in digital" is meant for Flex.  Should that remain or should it be changed?

mck1117, 2021-06-18:
IN_DIGITAL is a fine name, most digital sensors need a pullup anyway.

BeerMoneyMotorsports, 2021-06-19:
@Qwerty-OFF this sufficient to finalize?

Qwerty-OFF, 2021-06-19:
Yes, I have installed these pullups and pulldowns and proceed to the PCB. 
[image]

BeerMoneyMotorsports, 2021-06-21:
Im good with implementation and closing.

closed 2021-06-23 by rusefillc