rusefi/hellenNA6_issues #11: PCB tolerances

rusefillc, 2021-07-02:
What's the root cause? is that in the area where frame and module are merged?

[image]

rusefillc, 2022-01-07:
poor description in this issue I have no idea what it is about

@andreika-git @Qwerty-OFF @BeerMoneyMotorsports do we care or close?

Qwerty-OFF, 2022-01-07:
My mistake.  The track is located in the module.  This is not a problem here, but should not be done in the next rev.

rusefillc, 2022-01-07:
@Qwerty-OFF I need more info :( I am still missing context

Do we need a fix in hellen? what module specifically? please create a hellen ticket and mention here

Qwerty-OFF, 2022-01-07:
Not need fix it in hellen. Should be fixed in next revision of NA6.