rusefi/hellenNA6_issues #13: Hellen 64 yaml discrepencies

BeerMoneyMotorsports, 2021-07-14:
While testing the H64b board, it was discovered tachout was not working.  tachout changed pins on the stm32 which requires yaml updated.  If yaml is updated for h64b, it will break h64a.  Do we need to split the yamls between revisions?  Unsure on how to proceed.

rusefillc, 2021-07-14:
```

BeerMoneyMotorsports  10:47 AM
while testing the H64b, i noticed tach wasnt working.  i went ahead and bench tested since I have had this issue on every board so far.  Looking at the yaml, it appears that the pin for H64b changed for tachout.  with from PF12 to PI0.
10:47
this means i need to modify the yaml to suit but that will break the h64a tach out
10:47
is there a way to have both in the same yaml?  the header pin is the same.
10:48
this also means i am going back through the entire yaml and checking everything

BeerMoneyMotorsports  10:53 AM
it kinda looks like we will need seperate yaml between them

andrey  11:10 AM
@BeerMoneyMotorsports please create a ticket
11:10
I am very interested why the hell did pinout change

BeerMoneyMotorsports  11:13 AM
looks like the pin used for tach previously is now for boost control.  ticket in rusefi or Hellen NA6 issues?
```

rusefillc, 2021-07-14:
here is where this mine was placed:

https://github.com/rusefi/hellenNA6_issues/issues/6#issuecomment-861862674

BeerMoneyMotorsports, 2021-07-15:
Please see STM pin changes from Rev A to B

PIN-- RevA--RevB--Function
1E      PG4     PG9    CEL
3I       PF12    PI0     Tach
2J      -----     PF12   Boost control

Proposed Fixes
1E  Remove R39, Jump R39(header side) to P23?  Unsure if you can bypass chip M5B.
3I  Remove R24, Jump OUT_LOW_11 to R24(Header side)
2J unpopulated on H64a.  could jump R24(chip side) to pin 2J

BeerMoneyMotorsports, 2021-07-21:
Updated yaml for rev B pinout. Picture below shows jumper for tachout on rev A and latest yaml. No solution for CEL yet. 

tach out rev A-   Remove R24, Jump OUT_LOW_11 to R24(Header side)

[image]

BeerMoneyMotorsports, 2021-08-13:
Closing issue without CEL fix.  low priority at this point.