rusefi/hellenNA6_issues #14: Two channel aux DC motor driver

rusefillc, 2021-07-22:
our favorite ST chip is dual-channel right? let's wire both channels to just large pads for now

rusefillc, 2021-07-30:
So like https://github.com/rusefi/hellen88bmw-issues/issues/1 - two JST headers for TPS and pedal, and Molex MiniFit Jr. 4-pin? 

two pins of Molex MiniFit Jr. 2-pin for ETB and two other pins for aux H-bridge

BeerMoneyMotorsports, 2021-08-02:
@Qwerty-OFF add on etb driver and vertical board mounted connector. Connector requires 10a capacity.

rusefillc, 2021-08-09:
See BMW for 6 pin TPS and 4 pin ETB pinout

[image]

See BMW for MiniFit for ETB wires

only difference - let's have TWO separate MiniFit one for each DC motor channel

second DC motor let's wire control pins so that both channels are equally functional