rusefi/hellenNA6_issues #19: VSS

BeerMoneyMotorsports, 2021-11-11:
VSS is available on pin 1M for 94-95. Currently has a 5v supply which creates a short in 94/95.  This should be changed to a VSS Input.

rusefillc, 2021-11-11:
Thats a HW change? Do you have relevant of schematics could help for context

BeerMoneyMotorsports, 2021-11-11:
delete existing resistor and connection to +5v

[image]

add connection to VSS(example NB1 pin 2D)
VSS is already connected to STM32, just no connection to pin 1m. 

[image]

schematic showing connection from 1M to VSS

[image]
[image]

rusefillc, 2021-11-11:
Hold on, it's not a short. It's a pull-up.

Let's voice talk when you have time.

BeerMoneyMotorsports, 2021-11-11:
i have updated original problem statement to correct my mistake regarding a short.

BeerMoneyMotorsports, 2022-10-03:
This is fixed with Rev C.  closing.

closed 2022-10-03 by BeerMoneyMotorsports