rusefi/hellenNA6_issues #26: NA6 update to NB2 board format

BeerMoneyMotorsports, 2022-03-13:
Next revision of 90-95 will go to the same board shape as NB2.  Adding ETB and being consistent with the rest of the series of ECUs.  This will allow me to go to one case for all ecus.  

1- update to NB2 board shape
2- add ETB 
3- expand to 76 pin connector.  Use 4th connector for ETB.  
4- Add VSS connection to 1P (notate connection for 94-95 only.

BeerMoneyMotorsports, 2022-04-15:
5- Delete R12 and R17.  Not required for 90-93 batch injection.

BeerMoneyMotorsports, 2022-04-15:
@andreika-git do you have a timeframe of when you might be able to tackle this update?

andreika-git, 2022-04-15:
> @andreika-git do you have a timeframe of when you might be able to tackle this update?
Maybe after May 9, is it ok?

BeerMoneyMotorsports, 2022-05-19:
@andreika-git how is your schedule looking?  Id love the update to happen so i can have this revision for MATG.  That would require a mid June deadline.

andreika-git, 2022-06-17:
https://github.com/rusefi/hellen64_miataNA6_94/commit/e3706f638bb59a0f2b365d2dacf80c44caab64af

@BeerMoneyMotorsports Can you please review?

BeerMoneyMotorsports, 2022-06-22:
Can we maintain the Molex connector for ETB + and -?  

VSS input is currently going to 1P.  That should be going to 1M with R20 and R23 deleted.  VSS already has its own 5v pullup.

andreika-git, 2022-06-23:
[image]

@BeerMoneyMotorsports Can you please review, including the current BOM replacements?

andreika-git, 2022-06-23:
https://github.com/rusefi/hellen64_miataNA6_94/commit/c9e5d2dc44aef6e154a183a53b2d874076f72541

andreika-git, 2022-06-23:
Also, what about #28 and #29 ? Do we need them for rev.C?

BeerMoneyMotorsports, 2022-06-23:
#28 is needed
#29 was a jlc build issue.  No need for board change.

andreika-git, 2022-06-24:
Are we ready to go? What about BOM replacements?

BeerMoneyMotorsports, 2022-06-24:
Doing full review now. Ill update with more info later but i see an issue.  

[image]

NA6 uses the 176mcu like the NB2 and not the 144.

andreika-git, 2022-06-25:
> NA6 uses the 176mcu like the NB2 and not the 144.

Done!
[image]

andreika-git, 2022-06-28:
> BOM- Changes 1- U601 - C157715 

Done.

2- U490/491 - c50714 

Done.

3- Q601- none for this, as ill use the alternate footprint and stock of VND14N 

Done

4- R29,F101 - R29 shows to be a jumper and not a fuse? C269114(f101) Out of stock 

It's a fuse, yes. Done, used an alternative part C70072.

5- U6 - I do not see an alternate for the TLS115 in other BOM. I see a note that indicates that this may be optional? Note related to R29. 

Yes, if R29 is populated, then U6 is optional. Or you can remove R29 and solder TLS115 manually.

6- U105 still shows 144 pin?

Fixed with C33993.

BeerMoneyMotorsports, 2022-08-27:
done, great work @andreika-git

closed 2022-08-27 by BeerMoneyMotorsports