rusefi/hellenNA6_issues #27: Flex Fuel dropout

BeerMoneyMotorsports, 2022-05-02:
I had some issues with flex % today. I was on a 45 minute drive and it went from 75 to 0 and started running lean. It did this 5 times and every time a complete power cycle fixed it. After stopping for an hour or so, i went back home. It did it once in the first 5 minutes and never again. 

I now have logs of the occurrence.  
https://rusefi.com/online/view.php?log=779

[image]
[image]

Measured the input with the scope.  

[image]

[image]

There was no change in the signal when the fault occurred.  Signal does show intermittent noise at times during operation. Sensor maintains vbatt.

BeerMoneyMotorsports, 2022-05-02:
Flex input uses in_sens1 in the hellen input module which ultimately goes to GPIOF-3 on the STM32F4

BeerMoneyMotorsports, 2022-05-02:
@andreika-git @mck1117

mck1117, 2022-05-02:
tested with multiple ECUs, or just one?

BeerMoneyMotorsports, 2022-05-02:
@mck1117 just the one.  I can attempt to try another

mck1117, 2022-05-04:
you can also try older firmware since we've touched some of this stuff recently

BeerMoneyMotorsports, 2022-05-12:
I have been running on newer firmware(05/04/22) and driven it a good bit with zero problems.  So firmware seems to be responsible.

closed 2022-05-12 by mck1117