rusefi/hellenNA6_issues #31: 144 mcu

BeerMoneyMotorsports, 2022-08-16:
Please make optional files for using a 144 pin mcu. Relatively urgent request.

andreika-git, 2022-08-16:
What do you mean by "make optional files? Do we need a new rev.D for NA6?

BeerMoneyMotorsports, 2022-08-16:
I need to order NA6 boards with 144 pin mcu. Do you create an alternate Gerbers and BOM?  This should not require nee rev as board ID should take care of firmware differences.

rusefillc, 2022-08-16:
One revision is one set of schematics and gerber files. If we change MCU that really wounds like new revision.

@BeerMoneyMotorsports why do you care for revision to not change?

andreika-git, 2022-08-17:
I propose to do a new revision (rev.D)

BeerMoneyMotorsports, 2022-08-17:
My concern is customer interpretation of revision stability.  This is my only problem with a new rev already. 

If we are just changing the MCU module, **to me**, that is not a revision change.  I feel like it could be defined as rev C 144 and rev C 176

If it requires changes outside the MCU module, then we should absolutely do a rev D

mck1117, 2022-08-17:
every non-bom change should get a letter rev

rusefillc, 2022-08-17:
MCU change is a huge change with very wide implications. I hear the user perception concern but we have to go new letter.

andreika-git, 2022-08-24:
Please specify the exact LCSC# for the 144-pin MCU.

BeerMoneyMotorsports, 2022-08-24:
C19156

andreika-git, 2022-08-24:
https://github.com/rusefi/hellen64_miataNA6_94/commit/71c8cce6a1e1273e8a474768ccd534fe1441d84f

BeerMoneyMotorsports, 2022-08-26:
The pinouts match and schematics look good.  i have reviewed the pin changes for the MCU and only found two which is great.

I think we can move forward. 

Note- noted changes are 
1G- ign1 I8 to C13
3I- out tach I0 to G12

BeerMoneyMotorsports, 2022-10-02:
yaml tested for the pinout differences from 176 to 144.  it seems to work as in it will break those particular outputs when changed. I think we are OK to move forward @andreika-git

closed 2023-03-06 by BeerMoneyMotorsports