rusefi/hellenNA6_issues #32: "Blip" with trigger error on several boards

BeerMoneyMotorsports, 2022-08-27:
I have 3 reports of a "Blip" while driving.  The report is that the engine and tach die, then come back on if the car is in motion(restarts itself) but will die if not in motion.  This issue is started under NA6 due to the first one reporting the issue.  

Current reports-
NA6 firmware-- [rusEFI 2022.06.29.hellenNA6.1831869224]
NA8 96 firmware-- [rusEFI 2022.06.29.hellenNA8_96.4107948449]
NB1 firmware-- [rusEFI 2022.07.01.hellen-nb1.3773366088]

logs-
NA6- https://rusefi.com/online/view.php?log=952
96- https://rusefi.com/online/view.php?log=1004
nb1- incoming

[image]
[image]
[image]
[image]

rusefillc, 2022-08-27:
@BeerMoneyMotorsports please recomment snapshot firmware since we've added EXTI errors data logging - need to reproduce issue with that logging

@BeerMoneyMotorsports how many miles do you put on rusEFI yourself on an average week?

BeerMoneyMotorsports, 2022-08-28:
@rusefillc i will have them update to latest snapshot and report back.

@rusefillc somewhere between 50-100 miles a week

BeerMoneyMotorsports, 2022-09-04:
Here is one with snapshot. 58 second mark

https://rusefi.com/online/view.php?log=1022

droneperson42, 2022-09-04:
Her'es my log of it happeneing on a Hellen NB1. 662seconds in https://www.rusefi.com/online/view.php?log=1025

droneperson42, 2022-09-04:
Captured on video as well https://youtu.be/kvi-AP7doiM