rusefi/hellenNA6_issues #33: 6 pin header added to BOM and CPL

BeerMoneyMotorsports, 2022-10-03:
76 pin header has been ordered into JLC for soldering in production.  JLC has not provided part # yet.

[image]

https://www.verical.com/pd/te-connectivity-headers---pcb-receptacles-5-174915-7-2055967?utm_campaign=octopart_2022&utm_content=inv_listing&utm_currency=USD&utm_keyword=5-174915-7&utm_medium=aggregator&utm_source=octopart

BeerMoneyMotorsports, 2022-10-05:
Updated, Part number for header is JLCPCB Part #: C5184583

andreika-git, 2022-10-06:
I suspect that 5-174915-7 is partially incompatible with our 178780-1: leftmost and rightmost holes (big 3.5mm ones) are misaligned!
[image]
[image]

BeerMoneyMotorsports, 2022-10-06:
this is correct.  when i install them, i clip off the holes on the end.  This is due to parts availability.  The center two holes provide plenty of support.

BeerMoneyMotorsports, 2022-10-12:
@andreika-git also Q305-312, injector drivers, have a new part number listed below.

[image]

closed 2023-03-06 by BeerMoneyMotorsports