rusefi/hellenNA6_issues #34: Rev B CEL output not functional

Keith-N, 2022-11-04:
Hellen NA6 Rev b, using snapshot build 2022.11.03 

Check engine light is not properly blinking on boot, or enabling through the use of the 'Bench test' menu in TunerStudio.

Wiring was checked with a DMM and a known good bulb was installed.
Check engine pin 1E is selected as the output.

The output was tested at the connector using a DMM. When the ignition was on, the voltage at the pin was pulled up to battery voltage through the bulb as expected. The output was never switched to ground either on ECU power up or bench test command.

rusefillc, 2022-11-04:
@Keith-N plz paste a tune URL

Keith-N, 2022-11-04:
Here it is [https://rusefi.com/online/view.php?msq=1282](https://rusefi.com/online/view.php?msq=1282)