rusefi/hellenNA6_issues #35: Rev D LED pinout

BeerMoneyMotorsports, 2022-11-09:
During review between C and D, i missed that the LED pinout is different on the STM32F4.  See screenshots below highlighting the difference.  This results in a critical error.  

Rev C 176 pin
[image]

Rev D 144 pin
[image]

rusefillc, 2022-11-09:
My forum signature has http://rusefi.com/s/questions in it

http://rusefi.com/s/questions links to https://opensource.com/life/16/10/how-ask-technical-questions

https://opensource.com/life/16/10/how-ask-technical-questions says "If you're getting an error message or debugging information, make sure you include it verbatim."

BeerMoneyMotorsports, 2022-11-09:
[image]

rusefillc, 2022-11-09:
just pushed a hack which removes the validation from NA6 board

BeerMoneyMotorsports, 2022-11-09:
I see where i can change the LED pins for 2, 3 and 4 in Tuner studio.  I do not see it for LED 1.  Also, It will revert the pins i change back after power cycle.

rusefillc, 2022-11-09:
One issue per per one ticket please. I was assuming the focus here was critical error.

BeerMoneyMotorsports, 2022-11-09:
I confirmed that the error is fixed but that was a symptom of the actual problem.  I assumed you applied the Hack so i could continue validation while we figured out how to fix the LED pin assignment.

BeerMoneyMotorsports, 2022-11-13:
How can we fix the LED pin selection?  I dont see the ability to change LED1 at all and the other 3 wont save.

BeerMoneyMotorsports, 2022-11-13:
adding email screenshot here for consistency.  

[image]

BeerMoneyMotorsports, 2022-11-13:
Looks like I need two things to fix this.  There are probably smarter ways but here is what I thought. 
1- add selectable pin for LED-1 (red critical)
2- Not have the LED selections locked in firmware.  I'm assuming that this is what resets them after I manually change the selections for LEDs 2, 3 and 4.  

Other options?  alternate board identifier for the revD with 144 chip?

closed 2023-03-06 by BeerMoneyMotorsports